foss-for-mips-arc-processors/gcc #34: Override default heap size?

jstraus59, 2015-03-06:
I see in the linker scripts (e.g. arc-elf32-lib/ldscripts/arcelf.xn) the following:

    . = . + (DEFINED(__HEAP_SIZE) ? __HEAP_SIZE : (DEFINED(__DEFAULT_HEAP_SIZE) ? __DEFAULT_HEAP_SIZE : 20k))  ;

Which would seem to imply that you can specify a value for __HEAP_SIZE to override the default heap size. I would like to do this using the ld command line, but I cannot find anything in the ld manual telling how to define a value for a linker script variable.

Does anyone know if there is a way to do this, either from the ld command line or any other way?

Thanks

Jim Straus

anthony-kolesov, 2015-03-06:
Hi Jim,

If you call ld via gcc, then options are: `-Wl,--defsym=__DEFAULT_HEAP_SIZE=256m -Wl,--defsym=__DEFAULT_STACK_SIZE=1024m`. If you call ld directly, then I guess it is just needed to omit `-Wl,`s, but I haven't tried myself.

Anton

jstraus59, 2015-03-06:
Hi Anton,

Yep, that does the trick (FYI, it does work on ld directly as well). Not sure how I missed that in the manual. 

Thanks for the help!

Jim

closed 2015-03-06 by jstraus59