foss-for-mips-arc-processors/gcc #47: how to have FSDIV instruction supported by arc gcc

ardeal, 2015-10-13:
Hi Prof.

In my Makefile, I add the following options to eaable arc float HW instructions:
EXTRA_CFLAGS += -mcpu=ARCv2EM -mdiv-rem -mbarrel-shifter -mspfp

But float divsion, float to int, int to float and float comparison are not supported by arc gcc natively and correctly.
Can you tell me how to enable those 4 HW instructions?

claziss, 2015-10-13:
Hi,
-mspfp refers to old ARC FPX extensions and are not supported by new ARCv2 processors. A correct option line will be:
-mcpu=arcem -mdiv-rem -mfpu=fpus -mbarrel-shifter
That will confer u basic single precision floating point HW support. For more information about mfpu options please refer to our manual.
Please let us know if the documentation is missing information.

ardeal, 2015-10-13:
Hi 
My code and dessembled code is as follows. it seems that dessembled code is not correct to HW instruction(no fsdiv, fsmul, fsadd instructions are included)

float a=0.4563f, b = 0.32324f, c=0.0f, e = 0.0f;
    c=e+a*b;
    c = c*e;
    c = c/a;
    pr_info(LOG_MODULE_OPEN_CORE, "---------------%f\n",c);

4004293c:	fc 1c c8 b6             	st.a       fp,[sp,-4]
40042940:	2f 22 06 20             	sexh       r18,r0
40042944:	0c 70                   	mov_s      r0,0
40042946:	9b 43                   	mov_s      fp,sp
40042948:	68 45                   	mov_s      r13,r3
4004294a:	fe 1b 04 b0             	sth        r0,[fp,-2]
4004294e:	a1 c1                   	sub_s      sp,sp,4
40042950:	88 46                   	mov_s      r14,r4
40042952:	a8 47                   	mov_s      r15,r5
40042954:	2f 21 46 20             	sexh       r17,r1
40042958:	2f 20 86 20             	sexh       r16,r2
4004295c:	0e d9                   	mov_s      r1,14
4004295e:	c3 42 05 40 c4 24       	mov_s      r2,0x400524c4
40042964:	6c 70                   	mov_s      r3,0
40042966:	8c 70                   	mov_s      r4,0
40042968:	ae 0a af ff             	bl.d       40041c14 <log_printk>

ardeal, 2015-10-13:
Hi,
Do you have the document about options to enable hw instructions like
 -mfpu=fpus
?
Can you share the detailed document?

ardeal, 2015-10-13:
Hi Prof.
Currently, in my Makefile, I use ARCv2EM for my ARC version, but I am not sure whether it is exactly correct. is there a way to check the version of my ARC?

anthony-kolesov, 2015-10-13:
Hi @ardeal,

To figure out exact version of CPU you need to read IDENTITY register. See ARC v2 PRM on what is the contents of this register.

GCC manual in share/doc/gcc/gcc.pdf has a list of -mfpu= values and what do they mean. There is no exact list of what are the exact instructions added by each option, but that can be derived from ARC v2 PRM.

Can you also post full test case with all compiler options.

claziss, 2015-10-13:
Hi
As far as I see from the dissassembly, you don't pass any optimization level to the compiler. Please can u use -O2 (for example). The default gcc compiler optimization level is set to zero. Thanks.

ardeal, 2015-10-14:
Hi Anthony and Claziss,
Many thanks for your quick response!
My test C code and corresponding dessembled code is as follows.

after readding gcc.pdf, I used the following options:
-mcpu=arcem -mdiv-rem -mfpu=fpus_all -mbarrel-shifter

The dessembled code is correct to my expection, but the code cannot be run on my ARC. I got the following panic issue when I run the test code:
PANIC for 1 (dump located on 0xa8014000)
ARC BUILD CKSUM: 93e43ff9 
ARC BUILD VERSION: 00000001 
ARC PANIC - eret=0x40039162 - sp=0xa8013af0

-------------------------------------------------------------------------------------
Test C code:
   float a=0.4563f, b = 0.32324f, c=0.0f, e = 0.0f;
    a=i2f(ax) + a;
    b = i2f(ay) + b;
    c=e+a*b;
    c = c*e;
    c = c/a;
    c = c* ax;
    if(c>5.6f)
    	c=3.2f;
    pr_info(LOG_MODULE_OPEN_CORE, "---------------%f\n",c);

-------------------------------------------------------------------------------------
Corresponding dessembled code:
40042950:	2f 38 ac 03             	???        r44,r0,r14
40042954:	01 30 80 0f e9 3e 27 a0 	fsadd      r0,r0,0x3ee9a027
4004295c:	2f 39 2c 04             	???        r44,r1,r16
40042960:	48 36 82 10             	fcvt32     r2,r14,2
40042964:	01 31 81 0f a5 3e b7 7f 	fsadd      r1,r1,0x3ea57fb7
4004296c:	00 30 41 00             	fsmul      r1,r0,r1
40042970:	41 31 01 00             	fsadd      r1,r1,0
40042974:	40 31 01 00             	fsmul      r1,r1,0
40042978:	07 31 01 00             	fsdiv      r1,r1,r0
4004297c:	00 31 81 00             	fsmul      r1,r1,r2
40042980:	03 31 80 8f b3 40 33 33 	fscmp      r1,0x40b33333
40042988:	0a 20 80 0f 4c 40 cd cc 	mov        r0,0x404ccccd
40042990:	ba 0b 60 04             	bl.d       4004b548 __extendsfdf2

40042994:	ca 20 4c 00             	mov.le     r0,r1
40042998:	c3 42 05 40 b4 24       	mov_s      r2,0x400524b4
4004299e:	00 43                   	mov_s      r3,r0
400429a0:	20 44                   	mov_s      r4,r1
400429a2:	0c 72                   	mov_s      r0,2
400429a4:	76 0a af ff             	bl.d       40041c18 log_printk

ardeal, 2015-10-14:
If I use the following option line:
EXTRA_CFLAGS += -mcpu=ARCv2EM -mdiv-rem -mbarrel-shifter -mspfp
The generated dessembled code is as follows, and it can be correctly run on ARC.
40042958:	2f 3b 2c 00             	???        r44,r3,r0
4004295c:	01 33 81 0f e9 3e 27 a0 	fsadd      r1,r3,0x3ee9a027
40042964:	2f 38 2c 00             	???        r44,r0,r0
40042968:	01 30 80 0f a5 3e b7 7f 	fsadd      r0,r0,0x3ea57fb7
40042970:	00 31 00 00             	fsmul      r0,r1,r0
40042974:	41 30 00 00             	fsadd      r0,r0,0
40042978:	40 30 00 00             	fsmul      r0,r0,0
4004297c:	9e 08 40 04             	bl         4004b218 __divsf3

40042980:	08 46                   	mov_s      r14,r0
40042982:	ee 0b 60 04             	bl.d       4004b56c __floatsisf

40042986:	a1 40                   	mov_s      r0,r13
40042988:	00 36 0e 10             	fsmul      r14,r14,r0
4004298c:	0a 21 80 0f b3 40 33 33 	mov        r1,0x40b33333
40042994:	9e 0a 60 04             	bl.d       4004b430 __gtsf2

40042998:	c1 40                   	mov_s      r0,r14
4004299a:	14 70                   	cmp_s      r0,0
4004299c:	0a 20 80 0f 4c 40 cd cc 	mov        r0,0x404ccccd
400429a4:	7a 0d 60 04             	bl.d       4004b71c __extendsfdf2

400429a8:	ca 20 8c 03             	mov.le     r0,r14
400429ac:	c3 42 05 40 8c 26       	mov_s      r2,0x4005268c
400429b2:	00 43                   	mov_s      r3,r0
400429b4:	20 44                   	mov_s      r4,r1
400429b6:	0c 72                   	mov_s      r0,2
400429b8:	76 0a af ff             	bl.d       40041c2c log_printk

If I use the following option line:
EXTRA_CFLAGS += -mcpu=arcem -mdiv-rem -mfpu=fpus -mbarrel-shifter

the generated dessembled code is as follows, ant ARC will go panic:
40042938:	2f 3b 2c 00             	???        r44,r3,r0
4004293c:	01 33 81 0f e9 3e 27 a0 	fsadd      r1,r3,0x3ee9a027
40042944:	2f 38 2c 00             	???        r44,r0,r0
40042948:	01 30 80 0f a5 3e b7 7f 	fsadd      r0,r0,0x3ea57fb7
40042950:	00 31 00 00             	fsmul      r0,r1,r0
40042954:	41 30 00 00             	fsadd      r0,r0,0
40042958:	40 30 00 00             	fsmul      r0,r0,0
4004295c:	46 0f 00 04             	bl         4004b0a0 __divsf3

40042960:	48 35 83 10             	fcvt32     r3,r13,2
40042964:	00 30 c3 00             	fsmul      r3,r0,r3
40042968:	03 33 80 8f b3 40 33 33 	fscmp      r3,0x40b33333
40042970:	0a 20 80 0f 4c 40 cd cc 	mov        r0,0x404ccccd
40042978:	2e 0c 60 04             	bl.d       4004b5a4 __extendsfdf2

4004297c:	ca 20 cc 00             	mov.le     r0,r3
40042980:	c3 42 05 40 14 25       	mov_s      r2,0x40052514
40042986:	00 43                   	mov_s      r3,r0
40042988:	20 44                   	mov_s      r4,r1
4004298a:	0c 72                   	mov_s      r0,2
4004298c:	8a 0a af ff             	bl.d       40041c14 log_printk

According to my experiments, the result can be summarized as follows:
1) mspfp option use fsadd, fsmul and fssub which are provided by arc extension instructions. The code can be run correctly on ARC.
2) fpus option use fsadd, fsmul, fssub, fsdiv and fscmp which are provided by arc extension too. The code can not be run correctly on ARC.

my questions are:
1) mspfp option and fpus option use the same instruction set, why can the code generated by mspfp be run on ARC correctly but that generated by fpus cannot be run on ARC?
2) is there a way to have fsdiv, fscmp and fscvt32 generated by mspfp as well?
3) there are the following ARC extension instructions in manual:
DSP_FP_DIV
DSP_FP_FLT2I
DSP_FP_I2FLT
DSP_FP_CMP
DSP_FP_SQRT
FP_ADD
FP_SUB
FP_MUL

The last 3 instructions can be supported by ARC gcc natively, but the first 5 can not be supported by ARC gcc natively with option mspfp on my ARC. is there a way to have the first 5 supported by ARC gcc natively?

anthony-kolesov, 2015-10-14:
Hi @ardeal,

The error you refer to has ERET register set to 0x40039162, but the disassembly doesn't have instructions from this address. Also the code you pasted is not complete - I compiled it with some modification and run on nSIM, and it worked fine, but perhaps modifications that I had to do to make this code compilable also removed the source of error. Maybe it is in i2f function?

What is the hardware platform you are running on? Do you have a TCF file for the core you are using?

Can't answer you last three questions - those are in Claudiu domain :)

claziss, 2015-10-16:
At this moment, the gcc compiler doesn't support mixing FPX instructions with DSP ones.

closed 2015-10-16 by claziss