foss-for-mips-arc-processors/gcc #48: issue: Linker symbol _DATA __BEGIN__ not found

ardeal, 2015-10-16:
Hi,

I add the following  options to have libm.a linker in my project:
LIB_INCLUDE_DIR_arc += -L ${T}/external/gcc-arc-elf32/arc-elf32/lib/em/ -lm

in my c code,
#include "math.h"
sin, cos, fabs can be correctly called. when I call acos or acosf, I got the following error message:
/home/smb/crb/wearable_device_sw/external/gcc-arc-elf32/bin/arc-elf32-ld: Warning: gc-sections option ignored
/home/smb/crb/wearable_device_sw/external/gcc-arc-elf32/bin/arc-elf32-ld: Error: Linker symbol __SDATA_BEGIN__ not found
/home/smb/crb/wearable_device_sw/external/gcc-arc-elf32/bin/arc-elf32-ld: final link failed: Bad value

Can you help to tell me what wrong is with my options?

claziss, 2015-10-16:
It seems to me, you are not using a proper linker script file. Why don't u use the gcc driver to solve all the standard includes and library paths?

ardeal, 2015-10-16:
which linker script do you refer to? tell me it, and I will paste it for your to check.

how to use gcc driver to solve all the standard includes and library paths?

claziss, 2015-10-16:
I have this example:
```C
#include <stdio.h>
#include <math.h>

#define PI 3.1415926535897932384626433832795f

int main (void)
{
  volatile float x =  (1.1 * PI * 45 / 180) ;
  float result = sinf(x);
  printf("The sine of %f is %f (%f)\n", x, result, sinf(x));

  return 0;
}
```

To compile successfully I use the following command line:
```sh
arc-elf32-gcc -O1 t01.c -o t01.x -lm
```

However, you need to have your compiler in the PATH (e.g. <your_install>/bin) and the LD_LIBRARY_PATH to prepend the arc library folder (e.g. <your_install>/lib).

closed 2015-10-16 by claziss

ardeal, 2015-10-16:
Hi Claziss,
My lib locates:
./external/gcc-arc-elf32/arc-elf32/lib/em/
My GCC locates:
./external/gcc-arc-elf32/arc-elf32/bin/gcc

in my makefile, I have the following command:
LIB_INCLUDE_DIR_arc += -L ${T}/external/gcc-arc-elf32/arc-elf32/lib/em/ -lm

How to modify my command to have acos supported?

claziss, 2015-10-16:
Hi,
I do not know your setup. I've provided a general example how it is working above. Please try it out.

For general issues you can ask questions via our Solvnet portal. One of our engineers will help u.

For questions about acos: 
http://www.tutorialspoint.com/c_standard_library/c_function_acos.htm

For questions about how to use the gcc compiler (in general) here:
https://gcc.gnu.org/onlinedocs/gcc/
http://pages.cs.wisc.edu/~beechung/ref/gcc-intro.html

or google: how to use gcc

This page should be used to report arc-gcc compiler issues (e.g., compiler internal errors, faulty code generation etc.)

Thank you for ur understanding.