foss-for-mips-arc-processors/gcc #49: options to have only libm built and linked

ardeal, 2015-10-26:
Hi,
I have already downloaded arc-gcc source code from github.  I would like to rebuild arc-gcc with the following configurations:
1) For newlib in arc-gcc, I would like to have libc and libm built and linked fully.
2) For other code except for newlib, preserve it the default configuration as you release arc-gcc binary.
3) our arc is: arcem with FPX and EIA.
When I run configure command in newlib folder, what options should be used? Can you tell me the detailed options for my platform and requirement?

If I modify item 1) as:
               For newlib in arc-gcc, I would like to have libm  only built and linked fully.
When I run configure command in newlib folder , what options should be used? Can you also tell me the detailed options?

Thanks!

ardeal, 2015-10-27:
Hi,
Can you help to answer my questions?
Furthermore, your commands in examples sections of the following link don't work:
https://github.com/foss-for-synopsys-dwc-arc-processors/toolchain/blob/arc-4.8-R3/README.md

Can you help to check this?

anthony-kolesov, 2015-10-27:
README file you mention is for 4.8-R3 release - instructions are more than one year old, so they may be not correct for the current toolchain or contain errors that have been fixed in later releases. Besides, you didn't described what exactly does't work.

To build libc/libm manually you should refer to `newlib/newlib/README` file - build instruction of newlib for ARC are not any different from those that are for any other architecture.

ardeal, 2015-10-28:
Do you mean that the source code of newlib in arc gcc soucre code is the same as the newlib souce code in Rethat official release?
there isn't any configuration change for arc specifically in newlib, right?

iguryanov, 2015-10-28:
Hi, 

you have access to Synopsys Solvnet help and the case where we discuss same questions with you. 
This forum is mainly for open source community to report errors in our latest code. 

As your company do have direct support from our Synopsys, please use official channels for getting basic usage help. 

Regards,
Igor.

claziss, 2016-07-28:
This is not a gcc compiler, closing it down

closed 2016-07-28 by claziss