foss-for-mips-arc-processors/gcc #52: 64 bits operations issue

Jeff006, 2016-07-28:
I'm facing an issue with 64 bits operation giving a wrong result. Here is my configuration:
- arc em6 v2 with mpy_option 2
- here is my gcc version:  arc-elf32-gcc.exe (ARCompact/ARCv2 ISA elf32 toolchain 2015.12) 4.8.5
- no optimization (-O0)

I'm having wrong result when doing int64_t multiply by constant.

Here is the function having troubles, the line where computa:

void computeNintNfract(uint32_t freq, uint32_t *NINT, uint32_t *NFRACT){

	uint32_t Nint;
	int64_t Lresult;
	int32_t Result;

	// here we want to compute the nearest integer of (freq/16)/1000000
	Nint = freq / 8000000; // ((freq/16)*2)/1000000
	if (Nint & 0x1) Nint++;
	Nint = Nint >> 1;

	Lresult = (int64_t) ( freq - Nint * 16000000);
	// here we want to compute Lresult*2^19/8000000
	
	/* for signed 64 bits multiply debug */
	int64_t test = foo(Lresult,137439);	
	
	Lresult = Lresult * ( (int64_t)137439 );
	/* test != Lresult here */
	
	Result = (int32_t) (Lresult / 2097152);
	if (Result & 0x1) Result++;
	Result = Result >> 1;

	*NINT = Nint-4;
	*NFRACT = (uint32_t)Result;

}

claziss, 2016-07-28:
There is a slight difference between calling foo and doing the multiplication at large. When you call foo, the arguments are cast to 32bit, while the multiplication is done on both operands on full 64 bit. Hence, you have two types of multiplication, one which does 32x32->64 and the other which does 64x64->Truncate(64).

More over, if you try your program on x86 you will get the same "wrong" behavior. I see no issue with the compiler. If I am wrong, please let me know.

For your reference, here is my x86 test program:

```C
#include <stdio.h>
#include <stdint.h>

uint32_t freq = 907555556;

int64_t foo (int32_t b, int32_t c)
{
  return (int64_t)c * (int64_t)b;
}

/* void computeNintNfract(uint32_t freq, uint32_t *NINT, uint32_t *NFRACT) */

void main (void)
{
  uint32_t Nint;
  int64_t Lresult;
  int32_t Result;

  /* here we want to compute the nearest integer of (freq/16)/1000000 */
  Nint = freq / 8000000;
  if (Nint & 0x1)
    Nint++;
  Nint = Nint >> 1;

  Lresult = (int64_t) ( freq - Nint * 16000000);
  // here we want to compute Lresult*2^19/8000000

  /* for signed 64 bits multiply debug */
  int64_t test = foo(Lresult,137439);

  Lresult = Lresult * ( (int64_t)137439 );
  /* test != Lresult here */
  if (test != Lresult)
    printf ("FAIL\n");
  else
    printf ("PASS\n");

  /* Result = (int32_t) (Lresult / 2097152); */
  /* if (Result & 0x1) Result++; */
  /* Result = Result >> 1; */

  /* *NINT = Nint-4; */
  /* *NFRACT = (uint32_t)Result; */
}
```

closed 2016-07-28 by claziss

Jeff006, 2016-07-28:
thank you very much for your fast answer indeed I managed to have good result by properly casting the operand of multiply ie to 32x32->64