foss-for-mips-arc-processors/gcc #53: optimization for volatile

Captionst, 2016-08-25:
volatile unsigned int read_reg(unsigned int address)
{
          unsigned int ret;
         asm( “ld.di %0, [%1]" : : "r"(ret), "r"(val) );
         return ret;
}

void caller( unsigned int addr )
{
     volatile unsigned int val;
     do{ val = read_reg( addr ); }while( val & 0x1 );
     // something other.
}

compile above code with -O2, we may get code like this:
...
0000: ld.di r4, [r4,0]
0004: st r4, [sp]
0008: ld r3, [sp]
000c: bbit0 r3, 1, 0020
0010: .....
001c: b 0004
0020: ....

since both local variable val and function read_reg is declared as volatile, I think, code at 001c should be "b 0000", am I right?

claziss, 2016-08-25:
It looks to me u defined volatile the return value of the function, probably this is not what u intended to do.

Captionst, 2016-08-25:
thanks.

Yes. I intend to do that. I want to mean that the value stored in address is volatile, I want the CPU reads it again and again, to check its value.

claziss, 2016-08-25:
I understand this, but u told to the compiler that only what the function returns is volatile. How it is obtained is not volatile...

Captionst, 2016-08-25:
U mean that I should re-write read_reg function to get it's return value volatile ？
Yet, I guess when a voaltie value is assigned to another variable, the volatile attribute is forgotten.

claziss, 2016-08-25:
Exactly, u did something like:
```c
int a;
volatile int b;

b = a;
```

Captionst, 2016-08-25:
Yet, if I declare ret in function read_reg as volatile, I get the same result.

Captionst, 2016-08-25:
I mean, I have no way to have CPU repeatly load value in address and check its newest value.

claziss, 2016-08-25:
Did u declared ASM also volatile?

claziss, 2016-08-25:
Did u declare ASM also volatile?

Captionst, 2016-08-25:
Oh. I am sorry. this works:
unsigned int read_reg(unsigned int address)
{
unsigned int ret;
asm volatile ( “ld.di %0, [%1]" : : "r"(ret), "r"(val) );
return ret;
}

closed 2016-08-25 by Captionst