foss-for-mips-arc-processors/gcc #54: shall I use __attribute__((interrupt("ilink1")))?

Captionst, 2016-09-19:
a simple example:
void _gpio_handler()
{
unsigned int gpio_int;
gpio_int = read_reg(int_status_reg);
do_something_before_clear_int();
write_reg( int_clear_reg, gpio_int);
}

if I decleare _gpio_handler as __attribute((interupt("ilink1"))), disamble the compiled code, I find that the register blink is pushed into stack, but now poped, anything wrong with my use of it?

void _gpio_handler() __attribute__((interrupt("ilink1")));

disamble code as:
c0000980 <_gpio_handler>:
c0000980:	c0f1                	push_s	blink
c0000982:	1cf4 b008           	st.aw	r0,[sp,-12]
c0000986:	c541                	st_s	r13,[sp,0x4]
c0000988:	1600 780d 9850 0040 	ld.di	r13,[0x98500040]
c0000990:	1c08 3700           	st	sp,[sp,8]
c0000994:	0d3e 0420           	bl.d	c0008ed0 <do_something_before_clear_int>
c0000998:	d8ca                	mov_s	r0,0xca
c000099a:	1e00 7360 9850 004c 	st.di	r13,[0x9850004c]
c00009a2:	c000                	ld_s	r0,[sp,0]
c00009a4:	c501                	ld_s	r13,[sp,0x4]
c00009a6:	1408 301c           	ld	sp,[sp,8]
c00009aa:	2021 0740           	j.d	[ilink]
c00009ae:	c0a4                	add_s	sp,sp,0x10

Captionst, 2016-09-20:
oh. gcc version info:
Using built-in specs.
COLLECT_GCC=/opt/arc-elf32/bin/arc-elf32-gcc
COLLECT_LTO_WRAPPER=/opt/arc-elf32/bin/../libexec/gcc/arc-elf32/4.8.5/lto-wrapper
Target: arc-elf32
Configured with: /home/opensource/gcc/arc_toolchain/gcc/configure --target=arc-elf32 --with-cpu=arc600 --disable-multilib --with-pkgversion='ARCompact/ARCv2 ISA elf32 toolchain arc-2016.03' --with-bugurl=https://github.com/foss-for-synopsys-dwc-arc-processors/toolchain/issues --enable-fast-install=N/A --with-endian=little --disable-werror --enable-languages=c,c++ --disable-shared --disable-tls --disable-threads --prefix=/home/gtoolchain/arc-elf32 --with-gnu-as --with-gnu-ld --enable-sim --enable-sim-endian=no --with-newlib --with-headers=/home/toolchain/arc-elf32/arc-elf32/include --with-libs=/home/toolchain/arc-elf32/arc-elf32/lib --disable-libstdcxx-pch 'CFLAGS_FOR_TARGET=-ffunction-sections -fdata-sections '
Thread model: single
gcc version 4.8.5 (ARCompact/ARCv2 ISA elf32 toolchain arc-2016.03)

claziss, 2016-09-20:
Issue confirmed also in 6.x.

claziss, 2016-09-21:
Function epilogue is instructed to save/restore blink when in ISR as well. Patch added to 6.x and 4.8.x.

closed 2016-09-21 by claziss