foss-for-mips-arc-processors/gcc #57: question about objdump

Captionst, 2016-10-13:
binary code: 0x72d7 0x0000, 0x1000

run objdump get: (gcc 4.8.5 get this)

cmp_s r2, r0
b 0x20000

should be: (gcc 4.8.4 get this)
cmp_s r2, 0x1000

compiler version: gcc version 4.8.5 (ARCompact/ARCv2 ISA elf32 toolchain arc-2016.03)

claziss, 2016-10-13:
Issue present in binutils mainline as well:

>arc-elf32-objdump -d t01.o

t01.o:     file format elf32-littlearc

Disassembly of section .text:

00000000 <.text>:
   0:   72d7                    cmp_s   r2,LIMM
   2:   0000 1000               b       131072

>arc-elf32-objdump -v
GNU objdump (ARCompact/ARCv2 ISA elf32 toolchain arc-2016.09-eng015-3-gb9672e3) 2.27.51.20160929

Captionst, 2016-10-13:
Yet gcc 4.8.4 works fine.

claziss, 2016-10-13:
The LIMM indicator is different between ARCv1 and ARCv2 (i.e., 0x3E  Vs. 0x1E). The test is only failing in the case of ARCv1 cpus. Patch on the way.

claziss, 2016-10-13:
Patch proposed here:
https://sourceware.org/ml/binutils/2016-10/msg00115.html

Just a short remark, this issue should have been opened on the binutils repository not on gcc :).

closed 2016-10-13 by claziss