foss-for-mips-arc-processors/gcc #58: optimization of volatile

Captionst, 2016-10-13:
#define BASE_ADDR           (0x10000000)

#define set_reg(addr, val) (*((volatile unsigned int *)(addr))) = (unsigned int)(val)
#define write_reg(addr, val)  (*((unsigned int *)(addr))) = (unsigned int)(data)

int foo()
{
    int a, b = 0x12345678;

    a = *((unsigned int *)BASE_ADDR);

    if( a > 0x20000 )  set_reg( a, b );
    else  write_reg(a, b);

    return 0;
}

compile above code with option: -mno-volatile-cache -O2

get something like this:

ld r2, [0x10000000]
mov_s r0, 0
st.di 0x12345678, [r2, 0]
j_s [blink]

obviously, write_reg branch is optimized. that is not what i prefered.

gcc version 4.85. (ARCompact/ARCv2 ISA elf32 toolchain arc-2016.03)

claziss, 2016-10-13:
I am not sure I understand what do u want to achieve with the example. As far as I can see, the write_reg has nothing to do with volatile and, hence, will be removed.
Even if it is volatile, the set_addr and write_addr does the same thing (I suppose data is a typo).  So, there is no reason why to have two identical store instructions.
Trying your example for ARM leads to the same output like for ARC. Maybe the example is not what you wanted to share.

Captionst, 2016-10-14:
I have a register file to operate, and there is a copy in memory.
sometimes, I need to operate the register file directly, so, I need a "by-pass cache" operate.
sometimes, I just want to operate the memory copy, one by one, and eventrally, the memory copy is written to the register file burstly. the memory copy may be cached, so, I do not need  "volatile"

which one do i need to operate, is saved in the BASE_ADDR.

claziss, 2016-10-14:
The maybe this will help you:

```c
__extension__ static __inline void __attribute__ ((__always_inline__))
set_reg (unsigned int addr, unsigned int val)
{
  asm volatile ("st.di\t %0,[%1]"
		: /* No outputs.  */
		: "r" (val), "r" (addr)
		: "memory");
}
__extension__ static __inline void __attribute__ ((__always_inline__))
write_reg (unsigned int addr, unsigned int val)
{
  asm volatile ("st\t%0,[%1]"
		: /* No outputs.  */
		: "r" (val), "r" (addr)
		: "memory");
}

#define BASE_ADDR           (0x10000000)

int foo()
{
    int a, b = 0x12345678;

    a = *((unsigned int *)BASE_ADDR);

    if( a > 0x20000 )  set_reg( a, b );
    else  write_reg(a, b);

    return 0;
}
```

which will lead to:

```asm
00000000 <foo>:
   0:   1600 7002 1000 0000     ld      r2,[0x10000000]
   8:   43c3 1234 5678          mov_s   r3,0x12345678
   e:   0e15 7082 0002 0000     brlt.nt 0x20000,r2,20 <foo+0x20>
  16:   1a00 00c0               st      r3,[r2]
  1a:   7fe0                    j_s.d   [blink]
  1c:   700c                    mov_s   r0,0
  1e:   78e0                    nop_s
  20:   1a00 00e0               st.di   r3,[r2]
  24:   7fe0                    j_s.d   [blink]
  26:   700c                    mov_s   r0,0
```

Is this OK for u?

claziss, 2016-10-14:
As this is not a compiler error, closing it.

closed 2016-10-14 by claziss

Captionst, 2016-10-14:
thanks. I got it.

this is the same issue as: https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/issues/53

claziss, 2016-10-14:
Thought, I've seen it somewhere. You can make a header file with inline functions as indicated and use them across your project.