foss-for-mips-arc-processors/gcc #60: Incorrect code when using frame pointer

jurobystricky, 2016-12-27:
The GCC compiler generates incorrect code with "-fno-omit-frame-pointer". (I verified this by using
GCC 6.2.1 tag "arc-2016.09-release" and also by the latest GCC 6.3.0 commit 9e505f78b4f50d6a5b89c2271073ca956a18f3dc).
When compiling with the "-fno-omit-frame-pointer", any local variables saved on the stack will be accessed via the frame pointer before the stack has grown to include those addresses. As a result, any interrupt that comes in during the time when the stack has not yet grown will corrupt the local variables. The following code snippet illustrates the problem:

```
400354cc <hmc5883l_sample_fetch>:
400354cc:	c0f1                	push_s	blink
400354ce:	1cfc b348           	st.aw	r13,[sp,-4]
400354d2:	80a2                	ld_s	r13,[r0,0x8]
400354d4:	1cfc b6c8           	st.aw	fp,[sp,-4]
400354d8:	439b                	mov_s	fp,sp         // frame pointer set here
400354da:	40a5                	ld_s	r0,[r13,0]
400354dc:	4163                	mov_s	r1,fp
400354de:	734c                	mov_s	r2,3
400354e0:	19e1 808a           	stb.aw	r2,[r1,-31]
400354e4:	1be8 b040           	st	r1,[fp,-24]       // frame pointer used here
400354e8:	712c                	mov_s	r1,1
400354ea:	420c                	ld_s	r2,[r0,0x4]
400354ec:	1bec b040           	st	r1,[fp,-20]
400354f0:	702c                	mov_s	r1,0
400354f2:	1bf0 b042           	stb	r1,[fp,-16]
400354f6:	2342 3781           	sub	r1,fp,0x1e
400354fa:	1bf4 b040           	st	r1,[fp,-12]
400354fe:	762c                	mov_s	r1,6
40035500:	1bf8 b040           	st	r1,[fp,-8]
40035504:	d907                	mov_s	r1,0x7
40035506:	1bfc b042           	stb	r1,[fp,-4]
4003550a:	8281                	ld_s	r12,[r2,0x4]
4003550c:	c1a8                	sub_s	sp,sp,0x20  // stack pointer increased here!
...
```

claziss, 2016-12-29:
Please provide a preprocessed file together with compiler options used.

Thank you

jurobystricky, 2016-12-29:
Compiler options (without paths to include files):  
```
-nostdinc   -c -g -std=c99 -fno-asynchronous-unwind-tables -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -ffreestanding
-Os -fno-stack-protector -fno-omit-frame-pointer -ffunction-sections -fdata-sections -mcpu=quarkse_em -mno-sdata -fno-delete-null-pointer-checks
-Wno-unused-but-set-variable -fno-reorder-functions -fno-defer-pop -Wno-pointer-sign 
-fno-strict-overflow -Werror=implicit-int
```

jurobystricky, 2016-12-29:
[hmc5883l.zip](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/677734/hmc5883l.zip)
[hmc5883l-preprocess.txt](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/677736/hmc5883l-preprocess.txt)

claziss, 2016-12-30:
Received, I'll check this problem asap. Thank you.

claziss, 2017-01-04:
Patch [done](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/commit/cbd8e54244cd02bdcf4f1057be3ce96631f35ac3).

Please let me know if you see any other issue.

Thank you for signalizing this.

closed 2017-01-04 by claziss