foss-for-mips-arc-processors/gcc #63: -fno-omit-frame-pointer and save_return_addr

joabreu, 2017-03-02:
Hi,

I was trying to use stack unwinding using -fno-omit-frame-pointer option and I verified that, for some functions, the blink register is not saved into stack which makes it not possible to find the return address. I verified that arc.c file has a variable called save_return_addr which enables/disables the saving of the blink register. I believe this register should always be saved when we are using -fno-omit-frame-pointer option, can you confirm this? Adding an "|| !flag_omit_frame_pointer" to the assigning of save_return_addr variable makes all functions save the return address, which is what is needed to correctly do stack unwinding.

Best regards,
Jose Miguel Abreu

cupertinomiranda, 2017-03-02:
Hi Jose,

Our architecture does not always puts blink in the stack.
BLINK is only saved if any of the instructions in between the function epilogue and prologue, clobber it (i.e. changes it), or if the blink never needs to be restored (functions that never return).

Regarding unwinding we opted for a more optimal solution to create unwinding-tables in DWARF information (section .eh_frame). 
As not all applications require to (themselves) perform unwinding, this is not enabled by default and requires the use of -fasynchronous-unwind-tables option.

Nevertheless, unwinding in the current release has issues and will not always work if you do not use -fomit-frame-pointer.

I have recently worked in fixing the stack unwinder. I am sorry to say but doing what you suggested (as a fix) would be a cruelty to the tools.
The fix is much more complex. ;-)
Please expect a fix to the stack unwinding in the next release.

Hope it answers your questions.

Best regards (Atentamente),
Cupertino

cupertinomiranda, 2017-03-03:
Hi again,

Just to for the sake of resolution after our phone call.

The only problem with the workaround is that it starts to generate un-optimized code.
The push instruction both allocates space and stores the register value in the stack.
The instruction at 0x0a, allocates space in the stack for the upcoming store in 0x18.

Considering code correctness, this function could simply be :

lr %r2,[%status32]
or %r2,%r2,6
flag %r2
j_s [%blink]

It is generally preferable for size and speed to include some more complex unwinding mechanism (such as the one in libgcc), with the DWARF unwind-tables than un-optimizing the code by adding extra loads and stores to stack.

Cheers,
Cupertino

vineetgarc, 2017-03-10:
My 2 cents - frame pointer usage is generally an "unoptimization" and frowned upon more so for the sake of stack unwinding. In ARC architecture, by default we aggressively omit frame pointer and instead rely on tooling such as dwarf.

Now for various reasons/constraints user could still  want -fno-omit-frame-pointer to work - which is fair ! But the point is even if saving blink works for simple cases, it breaks when callee registers (r13-r25) are also involved and generated by compiler for a certain function - the reason is ABI. 

So, in a Frame Pointer based ISA, the activation record needs to be 
1. push return address (BLINK on ARC) on stack (SP increments)
2. push caller's stack base (FP on ARC) on stack (SP increments)
3. set this frame's base (FP) to current SP
4. push Callee regs on stack (r13-r25 on ARC) - actual regs depends on function code.
5. create space for any more local vars (increments SP)

However ARC ABI (historically) mandates function prolog to have steps 1, 4, 2, 3. With that FP no longer contains the base address of current call frame, thus we can't use it as a deterministic anchor to get BLINK/FP of previous frame. In absence of dwarf info, debuggers such as gdb and other tools work by disassembling code. This is obviously non trivial to do.

Long story short, FP can't be used for unwinding and saving BLINK alone is not sufficient either.

closed 2017-03-10 by cupertinomiranda