foss-for-mips-arc-processors/gcc #64: Is there a way to build libatomic for ARC processors?

jurobystricky, 2017-04-11:


claziss, 2017-04-13:
Hi, ARC cpus are providing atomic operation. You can enable them via -matomic when using the gcc compiler. If your processor doesn't support atomic operation, then, in theory, you should use libatomic which implement atomic ops. However, we didn't port this library.

claziss, 2017-04-26:
Hi,
I would like to know for which environment would you like to have the libatomic. We are considering to have it for linux like systems.
Thanks.

jurobystricky, 2017-04-26:
That sounds great.
We would use it with our baremetal ARC cross-compiler toolchain running under Linux.
Thanks

claziss, 2017-04-27:
just to clarify, the libatomic support will be added, in the first phase, for ARC linux which runs on cpus lacking load/store conditional instructions.

jurobystricky, 2017-04-27:
shouldn't the code also work (albeit possibly less efficiently) on CPUs that do have load/store conditional instructions?

claziss, 2017-04-27:
yes, it works, but what is the purpose for it. libatomic is specifically designed to emulate atomic ops which are required by gcc when these are not implemented/present in the target backend. Usually libatomic relays on pthread implementation (for posix systems) for its own internals.

claziss, 2017-05-05:
Hi,
I've made a [patch](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/commit/5b5eb4b51a351521b7e9541c4b81aae2a3f9a26a) which adds support for atomic ops via libatomic, also for bare-metal. Are you willing to give it a try?

N.B. To enable it you need to pass --enable-libatomic when configuring your toolchain. 

Thanks

claziss, 2017-06-13:
arc 2017.09 has libatomic enable for both elf and posix systems.

closed 2017-06-13 by claziss