foss-for-mips-arc-processors/gcc #65: how many registers shall I save for task/interrupt context

Captionst, 2017-07-11:
except the base register(r0~r31,loop_counter), which register shall I save for task/interrupt context, when dsp lib (Mul32x16/MDAC24x24) enabled.

in the dissamble code, I only find r57 is used, how about r58/r59?

thanks.

Captionst, 2017-07-12:
BTW: How should I restore r57? any difference between the following 2 ways?
ld.ab r13, [sp, +4]
mov r57, r13

and:

ld.ab r0, [sp, +4]
mov r57, r0

claziss, 2017-07-12:
You need to save/restore all the registers which you plan to touch in your interrupt. If you are not sure, then save all of the available registers existing on your architecture. If your architecture uses r58/r59 then save them as well.

N.B. For interrupts, gcc provides _interrupt_ function attribute. Maybe it helps you.

You should use a single instruction to save/restore an extension core register:
```asm
ld.ab r57,[sp,4]
```

closed 2017-07-12 by claziss

Captionst, 2017-07-13:
I am not sure, the document says, lp_count, which is r60, should NOT be the destination register of an LD operation; so, I think that "ld.ab r57, [sp, 4]" is illegal too. 

and , the program uses r13 as the mediate register to load r57 failed the test, yet r0 passed(maybe it will fail in the future.)

Captionst, 2017-07-13:
Yes. No extension core register can be the target of a load operation, including LD and EX.

claziss, 2017-07-13:
Indeed, the lp\_count register is special, and you need to take care using an intermediate register. However, if your architecture is ARC600 or ARCv2, then you can load directly into extension core registers.

If your architecture is ARC700 then you need to use a scratch register to perform your tasks (e.g. r12).

Captionst, 2017-07-14:
thanks. the architecture is ARC600 based.

Now, I get the DSP option reference document, it seems that I need to save more registers.

claziss, 2017-07-14:
Please check your documentation, it may be that some extension core regs to require extra moves via a regular register to save/restore them.

Thanks.

Captionst, 2017-07-17:
Thanks for your information.

the chip is built on ARC 600, with Mul32x16 dsp lib.
It seems that i need to save r56/r57, and 3 extra acumulator auxiliary registers.

BTW: if the mac24x24 is also built in. can I use the compile option -mmac-24 with -mmul32x16?

claziss, 2017-07-18:
Never try this combi :) U can, please let me know if you see issues.