foss-for-mips-arc-processors/gcc #66: Hit into __mulsi3 from 6.3.0/arcem/libgcc.a despite -mmpy-option=0 in Makefile

coco21, 2017-07-18:
We use 2017.03 toolchain on Windows host.

I checked TCF file and confirm on our ARCv2 EM6 design instance there is NO multiplier present (-mpy_option none).

I also checked in our Makefile we are well instructing the GCC compiler to NOT use hardware multiplier (-mmpy-option=0).

Now I see indeed there is a hit into the libgcc runtime using __mulsi3 routine that I think expects using EM6 ISA hardware support for multiplication despite we indicate no HW multiplier is present.

The execution context is the following :

- C Code : 
LocalVariable1   =   (  Define   +   (   ( localVariable2   -  1 ) * LocalVariable3;

- Corresponding assembly generated :
mov_s r0,r14
sub r1,r4,0x1
jl 0x6ecc         Boom here !  0x6eecc is __mulsi3 from 6.3.0/arcem/libgcc.a !!!
add_s r13,r0,0x01

Thx Christophe

anthony-kolesov, 2017-07-18:
What's the -mcpu value you are using?

coco21, 2017-07-18:
it's arcem

anthony-kolesov, 2017-07-18:
`-mcpu=arcem` pulls in libraries that include multiplier regardless of other -m options. You need to use `-mcpu=em4' http://embarc.org/toolchain/baremetal/gcc-mcpu.html#arc-em

coco21, 2017-07-18:
Ok, even if we use an EM6 one? I read the link you provide and understand there shall be no limitations toolchain vs. target vs. functionalities, am I right? Thx Christophe

anthony-kolesov, 2017-07-18:
Yes, you can use it for em6 as well. Unfortunately there is some discrepancy here with the hardware configurations, but is is the way it is. You need to select the `-mcpu` value that selects library that can run on your processor, then you can use individual `-m` options to fine-tune code generation for your core.

claziss, 2017-07-18:
Using an cpu option which includes a hardware feature, will enable that feature regardless of your disable options. I.e., if your arc template uses hw floating point support, then you cannot turn it off, it will always use it as all the libs and support functions are compiled using hw-floating point feature.
Thus, it is best to use a less feature reach cpu and pimp it up with options than using a full feature cpu and disabling the options.

But this leads me into thinking that may make sense to throw a warning or so, if one tries to disable an ALWAYS feature.

coco21, 2017-07-19:
Thanks for the guidance. Using the option em4 seems to work - we keep in our head to pimp factor in case we miss something.
Regarding the warning: yep this is something we were wondering why the toolchain didn't say anything when switching of an ALWAYS features - that would help. So you put that for 2017.09 release as well?

I will now close this ticket.

Christophe

claziss, 2017-07-19:
I'll keep it open, until I come with the patch :)

coco21, 2017-07-19:
Ok works for me ;)
Thx Christophe

claziss, 2017-10-18:
We crafted a patch which will warn the user if he/she attempts to turn off an architecture always-on option. However, we will not spur extra warnings if one tries to turn off an optional option such like the mpy in the given example. The reason is the extra warnings which will perturb other users in their work.
This new feature will be available in 2017.09 release.

closed 2017-10-18 by claziss