foss-for-mips-arc-processors/gcc #67: global / extern variable not working.

adaptrum-ver, 2017-11-21:
I have C code which is working well using Metaware. But when I want to compile using arc-gcc, I do not have any compilation error, but test execution hangs due to access points to unknown address. This indicate the assembly code generated by arc-gnu is not correct. 
Let me know if there is any option available to generate ELF file or any limitation in "arc-elf32-tcf-gcc" to use with extern variable or global variable.  
Here are the option using for C compilation
arc-elf32-tcf-gcc -nostdlib -nostartfiles -ffreestanding --tcf 1.tcf 1.s 1.c -o 1.elf -lc -Wl,-T 1.x

Since test is working well with Metaware, hence it is difficult to change the test, so looking for option in arc toolchain, if there is any.,

claziss, 2017-11-21:
Hi,

Without having an example, I can just make an educated guess. Thus, as far as I can see your global variable is placed in a custom assembly file. Usually, GCC uses small data accesses for the global variables. Hence, it may be you haven't placed that variable in the right segment, or your linker script is missing bits to place the external variable into the expected location.

Although, your example may work for MWDT, it doesn't mean it should also work for GCC :)

If you can share an example which is faulty, then I can help your further on.

Thank you,

adaptrum-ver, 2017-11-22:
Sorry I missed the example. Here is the example: 
In the below example err is declared globally, the execution hangs in ARC-GNU while it is working for Metaware. 

int err= 0;
int main(void)
{
    if(err==0)  pass;
    else fail;
}

int fail_condition_chk {
  if (fail_condition) err += 1;
}

claziss, 2017-11-22:
The above code snip will never compile in the first place due to missing pass/fail macros. Please provide the exact sources which are giving you hard time. Thank you

closed 2018-01-09 by claziss