foss-for-mips-arc-processors/gcc #68: undefined reference to __atomic*() functions while cross compiling usrsctp using arc 2017.09 toolchain

sampathkumar81293, 2018-01-17:
I am cross-compiling usrsctp for ``arc-linux-uclibc `` host on ubuntu 14.04. But it fails out with this error trace

``../usrsctplib/.libs/libusrsctp.so: undefined reference to `__atomic_compare_exchange_4' ``
`` ../usrsctplib/.libs/libusrsctp.so: undefined reference to `__atomic_fetch_add_4' ``
`` ../usrsctplib/.libs/libusrsctp.so: undefined reference to `__atomic_fetch_sub_4' ``

 I currently have ``arc 2017.09`` toolchain on my machine. how do I approach it?

Thanks

abrodkin, 2018-01-17:
Looks like you need to add `-matomic` flag to your cross-gcc command line.

sampathkumar81293, 2018-01-17:
Hi ,

I added the flag ``-matomic`` to the Makefile. But it throws the same undefined reference error. Please find the trace below

``libtool: link: arc-linux-uclibc-gcc -matomic -g -O2 -std=c99 -pthread -D_GNU_SOURCE -pedantic -Wall -Werror -g -O0 -o .libs/tsctp tsctp.o  ../usrsctplib/.libs/libusrsctp.so -pthread -Wl,-rpath -Wl,/home/tars/open_source_libs/lib ``
`` ../usrsctplib/.libs/libusrsctp.so: undefined reference to `__atomic_compare_exchange_4'`` 
``../usrsctplib/.libs/libusrsctp.so: undefined reference to `__atomic_fetch_add_4'``
``../usrsctplib/.libs/libusrsctp.so: undefined reference to `__atomic_fetch_sub_4'``

Thanks

abrodkin, 2018-01-17:
I'm wondering if you use pre-built ARC GNU tools from https://github.com/foss-for-synopsys-dwc-arc-processors/toolchain/releases/tag/arc-2017.09-release or build it yourself in Buildroot?

sampathkumar81293, 2018-01-17:
I built it myself on my machine

abrodkin, 2018-01-17:
Could you please elaborate a bit on how do you build it?
Do you use our build scripts or Buildroot?

sampathkumar81293, 2018-01-17:
Sorry for not providing the details.
I built it using the build_all.sh script  on Ubuntu 14.04 machine  from the  source.tar file  provided.

abrodkin, 2018-01-17:
The next question how do you execute your toolchain build?
I.e. your command line where you execute our scripts.

sampathkumar81293, 2018-01-17:
I used ./build-all.sh --no-elf32 --install-dir $INSTALL_ROOT  command to build  the toolchain. And added the  INSTALL/bin to my PATH

claziss, 2018-01-17:
Hi, you build the toolchain for ARC700. This architecture lacks atomic support.

sampathkumar81293, 2018-01-17:
Hi, is there a way I can mimic the functionality of __atomic*() functions using other implementations?

claziss, 2018-01-19:
If your A700 has the "memory operation" feature, then please pass -matomic when compiling. See my example:

```c
#define STRONG 0
#define WEAK 1
int v = 0;

int
atomic_compare_exchange_STRONG_RELEASE_ACQUIRE (int a, int b)
{
  return __atomic_compare_exchange (&v, &a, &b,
                                    STRONG, __ATOMIC_RELEASE,
                                    __ATOMIC_ACQUIRE);
}
```

and the result:

```asm
        .file   "t01.c"
        .cpu ARC700
        .arc_attribute Tag_ARC_PCS_config, 3
        .arc_attribute Tag_ARC_ABI_rf16, 0
        .arc_attribute Tag_ARC_ABI_pic, 0
        .arc_attribute Tag_ARC_ABI_tls, 1
        .arc_attribute Tag_ARC_ABI_sda, 0
        .arc_attribute Tag_ARC_ABI_exceptions, 1
        .section        .text
        .align 4
        .global atomic_compare_exchange_STRONG_RELEASE_ACQUIRE
        .type   atomic_compare_exchange_STRONG_RELEASE_ACQUIRE, @function
atomic_compare_exchange_STRONG_RELEASE_ACQUIRE:
        mov_s r2,@v     ;14
        .align 2
.L2:
        llock r3,[r2]
        cmp_s r3,r0
        bne_s @.L3
        scond r1,[r2]
        bne_s @.L2
        .align 2
.L3:
        mov_s r0,1      ;0
        j_s.d [blink]
        sub_s.ne r0,r0,r0
        .size   atomic_compare_exchange_STRONG_RELEASE_ACQUIRE, .-atomic_compare_exchange_STRONG_RELEASE_ACQUIRE
        .align 4
        .global atomic_compare_exchange_WEAK_RELEASE_ACQUIRE
        .type   atomic_compare_exchange_WEAK_RELEASE_ACQUIRE, @function
```

Otherwise, you need to use libatomic which should be already build.

closed 2018-01-23 by claziss