foss-for-mips-arc-processors/gcc #69: Invalid CFA information for delayslot instructions

cupertinomiranda, 2018-12-13:
The problem is that unwinder uses CFA information based on the address it is currently is at (return address from upper call).
Since delay slot instructions are executed out of order in respect to its address, i.e. instruction in delay slots get exectued before its counter-part delayslot flag instruction (parent instruction), however its address in the code is higher.

For simplification and validation, I provide a failing test case, which the unwinder destroys the stack due to this same issue.
[delay_slot_unwinder_problem.tar.gz](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/2676866/delay_slot_unwinder_problem.tar.gz)
In the extract of the assembly you can see cfa information for the "sub_s sp,sp,12" (in a delay slot) is placed below its instruction, associating it with the final address for that instruction instead of the branch.

\# arc-snps-linux-gnu-objdump -dr test3
221 0001098c <_Z4foo14data>:
222    1098c:       c0f1                    push_s  blink
223    1098e:       0ffa ffef               bl.d    -8      ;10984 <_Z4foo24data>
224    10992:       c1a3                    sub_s   sp,sp,0xc

\# arc-snps-linux-gnu-readelf --debug=frames-interp test3
000000d0 00000018 000000d4 FDE cie=00000000 pc=0001098c..00010994
   LOC   CFA      ra    
0001098c r28+0    u     
0001098e r28+4    c-4   
00010994 r28+16   c-4   ----> unwinder only uses this if return address would be higher then 10994.

One possible solution would be to change CFA information from delayslot instruction to parent instruction address.
We have to check what other similar targets do.

vineetgarc, 2018-12-13:
Can we also check what Metaware does here ?

vineetgarc, 2019-01-10:
Ping ?

claziss, 2019-01-16:
Fixed by commit 87554d7c3ad02ac8ae11634cb69d6097869b8dfd.

closed 2019-01-16 by claziss

abrodkin, 2019-01-16:
@claziss Could you please make sure it is pushed to upstream 8.x (so it gets in 8.3) and master (or 9.x whatever is the right name now) branches?

claziss, 2019-01-16:
Sure

abrodkin, 2019-01-16:
FWIW it does fix GDB at least in that case that failed immediately:
```
root@nsimhs:~# gdb /bin/busybox
GNU gdb (GDB) 8.0.50.20171026-git
...
and "show warranty" for details.
This GDB was configured as "arc-oe-linux".
Type "show configuration" for configuration details.
...
Reading symbols from /bin/busybox...(no debugging symbols found)...done.
(gdb) blah
Undefined command: "blah".  Try "help".
```