foss-for-mips-arc-processors/gcc #70: gcc built file not showing r58/r59 in Metaware debugger registers window

vineetgarc, 2019-02-20:
I've been debugging linux kernel / user code where gcc is freely using r58:r59 as scratch registers. However mdb doesn't seem to be detecting that's the case. I check with DavidP and it seems this only happens when the corresponding elf has following arcextmap sections which is added by Metaware linker. How can this be facilitated for GNU.

Content of section .arcextmap (#14) size=0xc6 start=0x37c8
     0: .extCoreRegister accl, 58, r|w
     8: .extCoreRegister acch, 59, r|w

claziss, 2019-02-21:
Unfortunately, we don't support extCoreRegister, only extAuxRegister. Anyhow, that it seems a bit shaggy to conditionalize dwarf debugging on the existence of arcextmap. A better approach will be to use DWARF extensions. However, for the problem in question, probably it is not needed at all any extension.

vineetgarc, 2019-02-28:
Is gcc generating (or can it if not already) an object attribute which the Metaware folks can also generate to signalize usage of r58:r59 ?

vineetgarc, 2019-03-01:
But maybe you are right, debugger should infer from cpu toggles whether the register is present or not w/o having to use a .extcoreRegister attribute.

claziss, 2019-11-12:
Is there something which I can do, I would like to close this issue as it is not GNU related. The third party tools needs to proper set their architecture info.

closed 2023-05-31 by claziss