foss-for-mips-arc-processors/gcc #71: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=89838

vineetgarc, 2019-03-27:
Ran into this when building the glibc test suite.

vineetgarc, 2019-03-27:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=89838 #71

claziss, 2019-03-29:
I'll look into asap. Is it only for gcc 8.x mainline?

vineetgarc, 2019-03-29:
I only checked gcc-8-branch @ commit 0d5ba57508c5 (not sure about mainline)
Also, a verification job with slightly later commit on gcc-8-branch seems to work, but still worthwhile to check.

abrodkin, 2019-03-29:
@vineetgarc @claziss may/should I mark this one for arc-2019.03 release?

claziss, 2019-03-29:
Sure, the issue is located somewhere in how tls is generated.

vineetgarc, 2019-03-29:
I didn't test with 2019.03, so please have Mikhail's job run glibc testsuite with 2019.03 based tools if affected or not !

abrodkin, 2019-03-29:
@vineetgarc well since this issue gets reproduced with both 8.x as well as master branch whatever fix happen will be applicable to our arc-2019.03 branch (and tag/release eventually).

claziss, 2019-04-01:
Patch proposed here 4c33b1cb6400341121700473da03d4083990a093

abrodkin, 2019-04-15:
@claziss is that the same one problem (got it while building Glibc with `-mcpu=hs38_linux`)
```
[ALL  ]      ../sysdeps/ieee754/flt-32/e_atan2f.c: In function '__ieee754_atan2f':
[ERROR]      ../sysdeps/ieee754/flt-32/e_atan2f.c:93:1: error: unrecognizable insn:
[ALL  ]       }
[ALL  ]       ^
[ALL  ]      (insn 154 153 155 25 (set (reg/v:SF 172 [ <retval> ])
[ALL  ]              (fma:SF (const_double:SF 3.0e+0 [0x0.cp+2])
[ALL  ]                  (const_double:SF 7.85398185253143310546875e-1 [0x0.c90fdbp+0])
[ALL  ]                  (reg:SF 58 r58))) "../sysdeps/ieee754/flt-32/e_atan2f.c":61 -1
[ALL  ]           (nil))
[ALL  ]      during RTL pass: vregs
[ERROR]      ../sysdeps/ieee754/flt-32/e_atan2f.c:93:1: internal compiler error: in extract_insn, at recog.c:2304
```

claziss, 2019-04-15:
@abrodkin the error shown is something different. Please make a new issue + relevant info on how to reproduce it. Thanks.

claziss, 2019-05-07:
Rebooted the patch with a new implementation a8aa243233e1493d78d762662b65a54cdb877533. Please recheck.

vineetgarc, 2019-05-20:
@claziss - is the patch merged upstream, i don't see it backported to gcc-8 ?

claziss, 2019-05-21:
@vineetgarc waiting for the nightly builds and validations. Until then, the patch is not qualified :)

vineetgarc, 2019-05-21:
@claziss - for 2019.03, the repo seems to have below, which seems different from the "new implementation"

2019-04-01 4c33b1cb6400 [ARC] Fix for PR89838.

abrodkin, 2019-05-22:
@claziss @vineetgarc the problem is we were not able to catch either issue in our automated flow running normal Linux verification as well as Glibc test-suite. Probably that's because we don't use `-mcpu=hs38_linux`. And since we cannot reproduce the failure we cannot conclude that proposed patch fixes it.

vineetgarc, 2019-05-22:
@abrodkin I think you are mixing 2 issues. I reported this issue when building glibc testsuite with standard -mcpu (not hs38_linux) and that issue remains today (on upstream gcc 8 as well as gcc 9)

I think u ran into something else which looked similar and perhaps that is specific to hs38_linux - consider opening a different issue to not add to confusion please.

claziss, 2019-05-22:
@vineetgarc indeed, the original patch is what I consider a hack on top of a hack, which is a sure receipt for disaster. Hence, I've come with a new implementation, which also cleans up some of the PIC implementation, though keeping the specs the same. The reason why I haven't pushed to 2018.03 branch is the imminent release of it.

abrodkin, 2019-05-28:
Got this reproduced in ARCJenkins `glibc_verification/257` job.
Job with only this change applied is in process in `glibc_verification/268` job.
This branch is used for that verification run - https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/tree/pr89838_fix.

abrodkin, 2019-05-29:
With ARCJenkins `glibc_verification/270` job proposed fix (https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/commit/4c33b1cb6400341121700473da03d4083990a093) was verified and this issue might be closed.

vineetgarc, 2019-05-29:
I propose we keep it open until the upstream bugzilla issue is resolved for better tracking !

abrodkin, 2019-05-29:
Ok, let's wait for upstream bug to be closed.
@claziss  who's going to close it here https://gcc.gnu.org/bugzilla/show_bug.cgi?id=89838?

claziss, 2019-06-06:
@abrodkin you used the wrong patch to solve this issue in https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/tree/pr89838_fix, please retry with 3b560dc7e88c07136fbcc4341461c0469674b1c8.

abrodkin, 2019-06-06:
@EvgeniiDidin could you please re-run glibc_verification/270 job with the right commit just mentioned by Claus?

EvgeniiDidin, 2019-06-06:
Hi @claziss,
Unfortunately I am not able to apply 3b560dc patch on top of upstream gcc(hunk3 fails), some nearby code in patch is not in upstream yet.
Could you please make this commit without dependency on previous ARC-specific changes?

claziss, 2019-11-12:
Closed as the patch is upstream.

closed 2019-11-12 by claziss