foss-for-mips-arc-processors/gcc #72: multibench printing wrong time value (glibc toolchain built with =mcpu=hs4x)

vineetgarc, 2019-03-28:
With a toolchain built with -mcpu=hs4x, saw a printing anomaly with multibench

```
[ARCLinux]# ./iDCT-4Mw1.out -v0 -w2
-  Info: Finished target initialization.
-  Info: Starting Run...
-- Workload:iDCT-4Mw1=228670582
-- iDCT-4Mw1:time(ns)=232468               <-- seed time value
-- iDCT-4Mw1:contexts=1
-- iDCT-4Mw1:workers=2
-- iDCT-4Mw1:iterations=100
-- iDCT-4Mw1:time(secs)=266.527954   <--- expected 232468/1000 = 232.468
-- iDCT-4Mw1:secs/workload=2.324680
```

This issue only happens with glibc toolchain built for this -mpcu: Not with uClibc toolchain with -mcpu=hs4x or even glibc toolchain built with standard -mcpu=hs38

vineetgarc, 2019-03-28:
This was a codegen issue due to missing cc clobber in gcc headers used in glibc
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=89877

gcc proposed fix posted: http://lists.infradead.org/pipermail/linux-snps-arc/2019-March/005590.html

Once merged in gcc, corresponding file needs to be sync'ed back to glicb as well.

claziss, 2019-03-29:
This is ok, but I am not sure I am the maintainer of that piece of code, moreover it is shared among multiple projects. I will include it to our branch, and I will fire an email to the gcc mailing list to see what is the way forward.

claziss, 2019-03-29:
Pushed locally 274a5a1d63fc5e4ea942165702363547e71525ac

vineetgarc, 2019-03-29:
Actually that code in under __arc__ so we could technically own it - but i understand the file overall is not ours so best to get advise from mailing list.

vineetgarc, 2019-04-04:
Patch merged in upstream gcc
glibc sync up patch posted : http://lists.infradead.org/pipermail/linux-snps-arc/2019-April/005654.html

claziss, 2019-04-08:
Patch merged into mainline & gcc8 branch.

closed 2019-04-08 by claziss

vineetgarc, 2020-03-19:
1. glibc patch reposted
https://sourceware.org/legacy-ml/libc-alpha/2020-03/msg00158.html

2. merged upstream glibc
2019-03-28 a9f4703d9341 gcc PR 89877: miscompilation due to missing cc clobber in longlong.h macros