foss-for-mips-arc-processors/gcc #73: Query about:  __attribute__  ((interrupt  ("ilink1"))) function attribute using

lipinggoke, 2019-04-13:
It seems that the GNU ARC GCC function attribute ' __attribute__  ((interrupt  ("ilink1"))) ' is different from MWDT '_Interrupt1' function attribute.

The GNU one doesn't save the register fields but the other one does.

This is the same interrupt routine code:

	typedef void (*isr_routine)(void);
	isr_routine will_trig_exception;

	 __attribute__  ((interrupt  ("ilink1"))) void isr_template(void)
	{
		will_trig_exception();
		return;
	}

assembler codes compiled by GNU ARC GCC  are these:

	00000258 <isr_template>:
	 258:	c0f1                	push_s	blink
	 25a:	1cfc b708           	st.aw	sp,[sp,-4]
	 25e:	c2e1                	push_s	r2
	 260:	1cfc b6c8           	st.aw	fp,[sp,-4]
	 264:	230a 3700           	mov	fp,sp
	 268:	78e0                	nop_s
	 26a:	12c2 b602           	ld.as	r2,[gp,-62]
	 26e:	7a40                	jl_s	[r2]
	 270:	78e0                	nop_s
	 272:	1404 341b           	ld.ab	fp,[sp,4]
	 276:	c2c1                	pop_s	r2
	 278:	1404 341c           	ld.ab	sp,[sp,4]
	 27c:	c0d1                	pop_s	blink
	 27e:	2020 0740           	j	[ilink]
	 282:	78e0                	nop_s

assembler codes compiled by MWDT  are these:

	isr_template:
	40041190: c4e1                     push_s	%r12
	40041192: c0e1                     push_s	%r0
	40041194: c1e1                     push_s	%r1
	40041196: c2e1                     push_s	%r2
	40041198: c3e1                     push_s	%r3
	4004119a: 1cfcb108                 push	%r4
	4004119e: 1cfcb148                 push	%r5
	400411a2: 1cfcb188                 push	%r6
	400411a6: 1cfcb1c8                 push	%r7
	400411aa: 1cfcb208                 push	%r8
	400411ae: 1cfcb248                 push	%r9
	400411b2: 1cfcb288                 push	%r10
	400411b6: 1cfcb2c8                 push	%r11
	400411ba: 1cfcbf08                 push	%lp_count
	400411be: 246a1080                 lr	%r12,[%lp_start]
	400411c2: c4e1                     push_s	%r12
	400411c4: 246a10c0                 lr	%r12,[%lp_end]
	400411c8: c4e1                     push_s	%r12
	400411ca: c0f1                     push_s	%blink
	400411cc: 70cf 10000000            mov_s	%r0,0x1000_0000		; will_trig_exception
	400411d2: 8000                     ld_s	%r0,[%r0]
	400411d4: 7840                     jl_s	[%r0]
	400411d6: c0d1                     pop_s	%blink
	400411d8: c4c1                     pop_s	%r12
	400411da: 246b10c0                 sr	%r12,[%lp_end]
	400411de: c4c1                     pop_s	%r12
	400411e0: 246b1080                 sr	%r12,[%lp_start]
	400411e4: c4c1                     pop_s	%r12
	400411e6: 749f                     mov_s	%lp_count,%r12
	400411e8: 1404340b                 pop	%r11
	400411ec: 1404340a                 pop	%r10
	400411f0: 14043409                 pop	%r9
	400411f4: 14043408                 pop	%r8
	400411f8: 14043407                 pop	%r7
	400411fc: 14043406                 pop	%r6
	40041200: 14043405                 pop	%r5
	40041204: 14043404                 pop	%r4
	40041208: c3c1                     pop_s	%r3
	4004120a: c2c1                     pop_s	%r2
	4004120c: c1c1                     pop_s	%r1
	4004120e: c0c1                     pop_s	%r0
	40041210: c4c1                     pop_s	%r12
	40041212: 242f003f                 rtie
	40041216: 78e0                     nop_s

The GNU ARC GCC even doesn't use the 'rtie' instruction to return back to the broken point.

In ARCompactISA, it says this:
"When the return from interrupt instruction, RTIE, is executed, A1 and A2 are loaded with values from the selected interrupt status link register (STATUS32_L1 for a level 1 interrupt or STATUS32_L2 for a level 2 interrupt). "

My Queries are:
1, Does the __attribute__  ((interrupt  ("ilink1"))) isr routine have to save the registers by itself while using GNU ARC Toolchain?
2, What registers in GNU ARC GCC compiler environment I should save and restore in the __attribute__  ((interrupt  ("ilink1"))) isr routine? Are the sames as MWDT doing?
3, The GNU ARC Toolchain doesn't use 'rtie' to return back to the broken point, so does it mean there should be some more operations needed to be done before using 'j_s [ilink1]' to go back ?

abrodkin, 2019-04-13:
@lipinggoke This issue is obviously related to GCC, not OpenOCD so move it to GCC repo.
@claziss Could you please take a look at this one?

lipinggoke, 2019-04-14:
I rechecked the ARCompactISA programmer's reference, and I think it should be use 'jal.f [ilink1]' instruction as an equivalent substitution to 'rtie' instruction, maybe not 'j [ilink1]' .

as for the registers to save/restore  in the entry/exit of interrupt, I 'grep'-ed the disassembler codes for extension core registers using. the result was like this:
[image]

Some instructions used r32~r59 range of extension core registers, but the object elf compiled by MWDT didn't (does this mean I have to save all the extension core registers in interrupt context? ), and also these instructions 'dsubh11, macdw, bdalc, drsubh21, xstw, orl, pkqb' are not in the ARCompactISA. would these instructions maybe cause instruction exception ? 

(PS: disassembler codes were dumped by 'arc-elf32-objdump -m arc700 -ald fw.elf > fw.D.S' )

My target core is ARC700. these are my CFLAGS options used for compiling:
CFLAGS = -ggdb -O2 -Os -mcpu=arc700 -fno-pic -mno-long-calls -nostdinc -nostdlib -mno-sdata -mlittle-endian -mswap -fgnu89-inline -fno-builtin -fno-common.

the contrast MWDT hcac CFLAGS options used for compiling are:
CFAGS = -arc700 -zallow_memory_overlap -Hnocpyr -g -O0 -Hnocplus -Hharvard -Hsdata0 -Hnocrt -Hoff=Behaved -Hon=Each_function_in_own_function -Xmpy

By comparison of 'mpy' optimization option between MWDT and  GNU ARC GCC, in GNU_Toolchain_for_ARC.pdf, as belowing:
[image]

none of number 3-6 for '-mmpy-option={3-6}' is suitable for arc700 compiling. error would be reported:
[image]

Only '-mmpy-option=0' is workable.
Why could MWD hcac use '-Xmpy', but GNU ARC GCC couldn't use the corresponding mpy optimization?

claziss, 2019-04-15:
Hi,
There are multiple questions here (the best is to open an issue for each), please let me try to address them in order:
1. About attribute **ilink1**, it seems it is not what you expect from it, and maybe we need to revisit it and make it more in the line of MWDT. But for this I need to get feedback from other customers. Meanwhile I would suggest to use attribute **naked** which will allow you to implement your own prologue/epilogue of a function as desired.
2. You get some exotic instructions in your code because you compiled it with **-Os** (for size) which will place jump-tables into code segment. The disassembler cannot discriminate between those jump tables and the real code and it will try disassemble it resulting in some gibberish output for that table only.
3. The ARC700 template is always using a multiplier. So you don't need to to anything to get it.

The gcc options are in gcc.pdf manual which you should have been receive it. In the section GCC Command Options -> Machine-Dependent Options -> ARC Options. Please let me know otherwise.

lipinggoke, 2019-04-26:
Thanks @claziss ,

After I refered to linux kernel  irq method of ARC cpu, this problem had been solved without using ilink1.
but I think, in embedded systems development, it would be more convient to use this attribute to automatically save the field registers as MWDT does, without additional assembler codes programming. 

Yet I found other user seemed to have the same query, at :
https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/issues/65
and you suggested to try interrupt function attribute.

I'd like close this issue and sincerely thanks for your guys warm-heated and patient support these days.

Regards.
-Li Ping

claziss, 2019-04-26:
Thank you Li Ping, but I wouldn't close this issue yet. I have discussed with my colleagues and we agreed to refurb the interrupt support for gnu.

claziss, 2019-05-17:
Hi,

Just crafted a first version which updates GNU support for interrupts here 163369380ec799858d50075926f098b7c3948651.
If everything ok, I will upstream it and back-porting it into gcc9.x branch. Once done, I will close this issue.

claziss, 2019-07-24:
A new way on how to interrupts is done. The patch is up streamed:
https://gcc.gnu.org/viewcvs/gcc?view=revision&revision=273761

closed 2019-07-24 by claziss