foss-for-mips-arc-processors/gcc #74: [glibc] HANGS: stdio-common/tfformat

cupertinomiranda, 2019-04-15:
[glibc] HANGS: stdio-common/tfformat

```C
#include <stdio.h>
#include <stdlib.h>

int main() {
    double d;
    const char *number = "-10247366524.086269378662109375";
    char *next;
    d = strtod(number, &next);
    //sscanf("-10247366524.086269378662109375", "%lg", &d);

    return (int) d;
}
```

claziss, 2019-04-17:
Hi, I need more info on this issue. Please can you provide:
1. the toolchain version were I can observe this issue;
2. the command option used;
3. other info on how to get this error duplicated.

claziss, 2019-04-18:
On a archs linux I execute tfformat_c.c:
```
# ./tfformat_c.x 
 *** main@57
 *** main@59
 *** main@67
 *** main@70: buffer is -102383357608906490000000000
 *** main@72
Error in line 15.  String is "-102383357608906490000000000", value is -1.02383e+26.
Success in line 15.  String is "-102383357608906490000000000", value is -1.02383e+26.
 *** main@57
 *** main@59
 *** main@67
 *** main@70: buffer is -10247366524.086269
 *** main@72
Success in line 16.  String is "-10247366524.086269", value is -1.02474e+10.
Error in line 97 using "%.100g".  Result is "1.7763568394002503e-15"; should be: "1.7763568394002504646778106689453125e-15".
Encountered 2 errors in 2 tests.
```

EvgeniiDidin, 2019-04-18:
Hi Claudiu,

Could you please provide more information about your setup?

I ran different configurations on hs_nsim and test fails only if full arc-2019.03-rc1 toolchain is used(GCC+binutils+glibc). With upstream GCC 8.3.0 or uclibc test runs well. I used buildroot to build everything.

claziss, 2019-04-18:
Please notice, the test is still erroneous, but it may have to do with something else. I used, arc-2019.03 latest branch + uclibc, compiled statically `arc-linux-gcc -O2 -static tfformat_c.c` run on qemu linux (glibc) simulation (image given on request).

EvgeniiDidin, 2019-04-18:
In my setup with uclibc test is erroneous as well. But building with arc-2019.03-rc1-glibc toolchain leads to test hang. Compiled statically "arc-linux-gcc tfformat.c"

Could you please try this setup?

vineetgarc, 2019-05-21:
so presume this will still be issue in released 2019.03

vineetgarc, 2019-07-09:
For reference:
STAR: 9001397900- [glibc] HANGS: stdio-common/tfformat
STAR: 9001397897- [glibc] FAIL: stdlib/tst-strtod-round

vineetgarc, 2019-11-07:
@EvgeniiDidin @falaleevms - do we know what the status of this issue is with latest upstream gcc etc.

EvgeniiDidin, 2019-11-08:
I ran tests  provided from STAR 9001397900 and this issue which were built with arc-2019-09-rc1 toolchain and gcc-9.2.0(binutils/glibc were arc-2019-09-rc1). There were no hangs.
I guess we can close this issue.

abrodkin, 2019-11-08:
> I ran tests provided from STAR 9001397900 and this issue which were built with arc-2019-09-rc1 toolchain and gcc-9.2.0(binutils/glibc were arc-2019-09-rc1). There were no hangs.

@vineetgarc @EvgeniiDidin @claziss What I really don't like here we never got to the bottom of this issue and so we don't know real root cause and given previously that issue was known to disappear due to completely unrelated commit I would expect it to re-appear anytime later with yet another unlucky commit, different configuration etc. So it would be good to understand what was the real root cause and then check if it was cured or not.

closed 2021-06-16 by vineetgarc