foss-for-mips-arc-processors/gcc #75: [BUG] ICE building e_atan2f.c: In function '__ieee754_atan2f' for -O2 -mcpu=hs38_linux -frounding-math

abrodkin, 2019-04-15:
```
# arc-unknown-linux-gnu-gcc -mcpu=hs38_linux e_atan2f.i -c -O2 -frounding-math

[e_atan2f.i.gz](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/3080892/e_atan2f.i.gz)
[e_atan2f.i.gz](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/3080894/e_atan2f.i.gz)

../sysdeps/ieee754/flt-32/e_atan2f.c: In function '__ieee754_atan2f':
../sysdeps/ieee754/flt-32/e_atan2f.c:93:1: error: unrecognizable insn:
 }
 ^
(insn 112 111 113 25 (set (reg/v:SF 172 [ <retval> ])
        (fma:SF (const_double:SF 3.0e+0 [0x0.cp+2])
            (const_double:SF 7.85398185253143310546875e-1 [0x0.c90fdbp+0])
            (reg:SF 58 r58))) "../sysdeps/ieee754/flt-32/e_atan2f.c":61 -1
     (nil))
during RTL pass: vregs
../sysdeps/ieee754/flt-32/e_atan2f.c:93:1: internal compiler error: in extract_insn, at recog.c:2304
0x55ed94 _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
        /SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/src/gcc/gcc/rtl-error.c:108
0x55edb0 _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
        /SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/src/gcc/gcc/rtl-error.c:116
0x9aa5c7 extract_insn(rtx_insn*)
        /SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/src/gcc/gcc/recog.c:2304
0x7c258d instantiate_virtual_regs_in_insn
        /SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/src/gcc/gcc/function.c:1599
0x7c258d instantiate_virtual_regs
        /SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/src/gcc/gcc/function.c:1969
0x7c258d execute
        /SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/src/gcc/gcc/function.c:2018
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
```

This is with vanilla GCC 8.3.0 built by Crosstool-NG:
```
.build/arc-unknown-linux-gnu/buildtools/bin/arc-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=/SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/buildtools/bin/arc-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/buildtools/libexec/gcc/arc-unknown-linux-gnu/8.3.0/lto-wrapper
Target: arc-unknown-linux-gnu
Configured with: /SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/src/gcc/configure --build=x86_64-build_pc-linux-gnu --host=x86_64-build_pc-linux-gnu --target=arc-unknown-linux-gnu --prefix=/SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/buildtools --with-local-prefix=/u/abrodkin/x-tools/arc-unknown-linux-gnu/arc-unknown-linux-gnu/sysroot --with-sysroot=/u/abrodkin/x-tools/arc-unknown-linux-gnu/arc-unknown-linux-gnu/sysroot --enable-shared --with-cpu=archs --with-pkgversion='crosstool-NG 1.24.0.1-fee743c' --enable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-libmpx --disable-libssp --disable-libquadmath --disable-libquadmath-support --with-gmp=/SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/buildtools --with-mpfr=/SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/buildtools --with-mpc=/SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/buildtools --with-isl=/SCRATCH/abrodkin/Projects/sources/git/crosstool-ng/.build/arc-unknown-linux-gnu/buildtools --enable-lto --enable-target-optspace --with-glibc-version=2.29 --disable-nls --enable-multiarch --enable-languages=c
Thread model: posix
gcc version 8.3.0 (crosstool-NG 1.24.0.1-fee743c)
```

abrodkin, 2019-04-15:
[e_atan2f.i.gz](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/3080895/e_atan2f.i.gz)

abrodkin, 2019-04-15:
Doesn't happen with arc-2019.03-rc1.

abrodkin, 2019-04-15:
Bug in GCC's bugzilla - https://gcc.gnu.org/bugzilla/show_bug.cgi?id=90103

claziss, 2019-04-16:
This is an gcc-8.x issue (backporting). I will close this thread as the Bugzilla issue being sufficient.

closed 2019-04-16 by claziss