foss-for-mips-arc-processors/gcc #77: Revision 272645 on top of 9.1.0 caused ICE

EvgeniiDidin, 2019-08-13:
After applying changes from revision=272645 on top of 9.1.0 ICE appeared.

---------------------------COMPILER CONFIGURATION------------------------------
Using built-in specs.

Target: arc-buildroot-linux-uclibc
Configured with: ./configure --prefix=/worktrees/br_ica/output/host --sysconfdir=/worktrees/br_ica/output/host/etc --enable-static --target=arc-buildroot-linux-uclibc --with-sysroot=/worktrees/br_ica/output/host/arc-buildroot-linux-uclibc/sysroot --enable-__cxa_atexit --with-gnu-ld --disable-libssp --disable-multilib --disable-decimal-float --with-gmp=/worktrees/br_ica/output/host --with-mpc=/worktrees/br_ica/output/host --with-mpfr=/worktrees/br_ica/output/host --with-pkgversion='Buildroot 2019.08-git-00654-gcf9e8ebe47-dirty' --with-bugurl=http://bugs.buildroot.net/ --disable-libquadmath --disable-libsanitizer --enable-tls --disable-libmudflap --enable-threads --without-isl --without-cloog --with-cpu=archs --enable-languages=c,c++ --with-build-time-tools=/worktrees/br_ica/output/host/arc-buildroot-linux-uclibc/bin --enable-shared --disable-libgomp
Thread model: posix
gcc version 9.1.0 (Buildroot 2019.08-git-00654-gcf9e8ebe47-dirty) 

------------------------REDUCED TEST CASE----------------------------------
char *XkbIndentText(unsigned size)
{
    static char buf[32];
    register int i;
    if (size > 31)
        size = 31;
    for (i = 0; i < size; i++) {
       buf[i] = ' ';
    }
    buf[size] = '\0';
    return buf;
}

------------------------COMMAND LINE USED FOR COMPILATION---------------------

./arc-linux-gcc -c test.c -fpic -O2
test.c: In function ‘XkbIndentText’:
test.c:14:1: error: unrecognizable insn:
   14 | }
      | ^
(insn 68 64 69 4 (set (reg:SI 178)
        (unspec:SI [
                (symbol_ref:SI ("buf.1422") [flags 0x2] <var_decl 0x7f068fa67240 buf>)
            ] ARC_UNSPEC_GOTOFFPC)) -1
     (nil))
during RTL pass: sched1
test.c:14:1: internal compiler error: in extract_insn, at recog.c:2310

EvgeniiDidin, 2019-08-13:
The error reproduce on latest GCC 9.2.0 release without any additional patches.

claziss, 2019-08-13:
Can u please tell me more about revision 272645? What is it, where is it.

EvgeniiDidin, 2019-08-13:
The revision is: 
https://gcc.gnu.org/viewcvs/gcc?view=revision&revision=272645
Is equal to https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/commit/228efbbb2f1d59ca7fc0ec09d4be75ce58fa583b

abrodkin, 2019-08-21:
@claziss any updates?

claziss, 2019-10-02:
The issue was due to a complex pic address which needs to be resolved using simpler logic. During this, we were losing the "constant" qualifier for the pic symbol reference, leading to the observed ICE.

Fix in 3eff7d38a0c9a618036b1b950f319bac689eb766.

closed 2019-10-02 by claziss

abrodkin, 2019-10-03:
@claziss could you please back-port it to the stable branch in upstream?

claziss, 2019-10-03:
@abrodkin are there any regressions? I am waiting for the nightly linux & elf :)