foss-for-mips-arc-processors/gcc #78: Invalid instruction is generated for -mcpu=em_mini -mcode-density

anthony-kolesov, 2019-08-14:
Test case:

```c
typedef long long uint64_t;

uint64_t f1(void)
{
        return 1;
}

void f2(void)
{
        uint64_t start_us = 0;
        while ((f1() - start_us) < 2);
}
```

Compile with:

```
$ arc-elf32-gcc --version
arc-elf32-gcc (ARCompact/ARCv2 ISA elf32 toolchain 2019.03) 8.3.1 20190225
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ arc-elf32-gcc -mcpu=em_mini -mcode-density -nostdinc -c test.c
/tmp/cc9irjja.s: Assembler messages:
/tmp/cc9irjja.s:43: Error: inappropriate arguments for opcode 'sub_s'
```

abrodkin, 2019-08-21:
@claziss could you please take a look at this one?

shahab-vahedi, 2019-08-27:
the problem seems to be rooted in the fact that `gcc` driver invokes the assembler without the `-mcode-density`:
```
$(AS) -mcpu=arcem -o /tmp/ccEZA0R2.o /tmp/cc9irjja.s
```

adding `-mcode-density` to the command above solved the issue:
```
$(AS) -mcpu=arcem -mcode-density -o /tmp/ccEZA0R2.o /tmp/cc9irjja.s
```

shahab-vahedi, 2019-08-28:
This has been fixed in [ef8b9fe](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/commit/ef8b9fe2d5dc4c477c87d020aee88506533aea5f) pushed to `arc-2019.09`.

closed 2019-08-28 by shahab-vahedi

shahab-vahedi, 2019-08-28:
also pushed to `dev` ([0c7dcc1](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/commit/0c7dcc115aaf07064a3878bca9542d711c41ee07)).