foss-for-mips-arc-processors/gcc #79: Building Glibc with -Og Fails

shahab-vahedi, 2019-10-07:
Having _toolchain script_ configured with:

```

shahab-vahedi, 2019-10-07:
problematic assembly (1):
```
.LVL539:
	.loc 5 3167 11 view .LVU1733
	extb.f	r0,r0
	bne.d @.L422
	mov_s.ne r0,12
	b_s @.L422
	.align 4
```

problematic assembly (2):
```
.LVL1396:
	.loc 4 1531 11 view .LVU4231
	extb.f	r0,r0
	bne.d @.L1021
	mov_s.ne r0,12
	b_s @.L1021
.LVL1397:
	.align 4
```

shahab-vahedi, 2019-10-07:
[reduced](https://embed.cs.utah.edu/creduce/) C code:
```
typedef struct
{
  struct
  {
    int length
  } table
} room;

struct house
{
  room *r
};

undefined;

_Bool bar();

func(struct house *h, int i, int whatever)
{
  for (;;)
  {
    _Bool a;
    if (h && h->r[i].table.length == undefined)
    {
      if (whatever)
      {
        a = bar();
        if (__builtin_expect(!a, 0))
          return 7;
      }
      break;
    }
  }
}
```

compile with:
```
cc1 test.c -mcpu=archs -Og -o test.s
as test.s -o test.o
```

output:
```
test.s: Assembler messages:
test.s:37: Error: Insn b has an instruction mov_s with limm in its delay slot.
```

problem:
```
bne.d @.L1      # 33    [c=20 l=4]  *branch_insn
mov_s.ne r0,7   # 35    [c=0 l=6]  *movsi_ne/2
```

shahab-vahedi, 2019-10-09:
`-fno-delayed-branch` by passes this problem.

claziss, 2019-10-09:
The issue is due to an mov instruction which is placed into a annul-true delay slot. Hence, it needs to be predicated. As it is having an immediate, it ends up in selecting `mov_s.ne` instruction of the `movsi_ne` pattern. However, this pattern choices list is ordered against what internal attribute `cond` assumes, hence, leading to the selection of the `mov_s.ne` mnemonic instead of the `mov.ne` mnemonic. Fix in 53803396251d58fd664b9fba8c30399972908756.

shahab-vahedi, 2019-10-09:
I confirm: it is fixed like a charm :)

closed 2019-10-09 by shahab-vahedi