foss-for-mips-arc-processors/gcc #81: Build uclibc linux toolchain with mainline gcc failed with ECR

falaleevms, 2019-11-05:
Building toolchain with upstream gcc (b7ede04df31ac4406cafdde3ad152b0c00869166) failed on configurations arc-arc700-linux-uclibc_arc700 and arceb-arc700-linux-uclibc_arc700 with following error:

----------&----------
/slowfs/ru20arcgnu1/gnu_verification/arceb-arc700-linux-uclibc_arc700/gcc/libstdc++-v3/libsupc++/eh_personality.cc: In function '_Unwind_Reason_Code __cxxabiv1::__gxx_personality_v0(int, _Unwind_Action, _Unwind_Exception_Class, _Unwind_Exception*, _Unwind_Context*)':
/slowfs/ru20arcgnu1/gnu_verification/arceb-arc700-linux-uclibc_arc700/gcc/libstdc++-v3/libsupc++/eh_personality.cc:731:1: internal compiler error: in lra_assign, at lra-assigns.c:1647
  731 | }
      | ^
0xc2deb5 lra_assign(bool&)
                /slowfs/ru20arcgnu1/gnu_verification/arceb-arc700-linux-uclibc_arc700/gcc/gcc/lra-assigns.c:1647
0xc28514 lra(_IO_FILE*)
                /slowfs/ru20arcgnu1/gnu_verification/arceb-arc700-linux-uclibc_arc700/gcc/gcc/lra.c:2482
0xbdb551 do_reload
                /slowfs/ru20arcgnu1/gnu_verification/arceb-arc700-linux-uclibc_arc700/gcc/gcc/ira.c:5511
0xbdb551 execute
                /slowfs/ru20arcgnu1/gnu_verification/arceb-arc700-linux-uclibc_arc700/gcc/gcc/ira.c:5697
----------&----------

claziss, 2019-11-05:
Confirmed.

claziss, 2019-11-06:
Reduced test:

```c
namespace a {
class b {};
} // namespace a
namespace std {
class type_info {};
void c();
} // namespace std
typedef unsigned m;
typedef signed d;
typedef unsigned e;
typedef unsigned f;
typedef enum {} g;
typedef int h;
void i(struct j *, int, m);
typedef long k;
typedef long l;
char aa(unsigned char *, l *);
char n(unsigned char *, k *);
static unsigned char *q(char ab, e, unsigned char *o, e *p) {
  union *s;
  f ac;
  switch (ab) {
  case 0:
    ac = (f)s;
  case 1:
    l ad;
    aa(o, &ad);
  case 9:
  case 2:;
  }
  *p = ac;
  return o;
}
struct r {
  e ae;
  unsigned char *u;
  char v;
};
static std::type_info *t(r *af, l) {
  e ag;
  q(af->v, af->ae, af->u, &ag);
  return reinterpret_cast<std::type_info *>(ag);
}
typedef const std::type_info x;
bool w(std::type_info *, const std::type_info *, void *);
bool y(r *, d);
namespace a {
g ah(h actions, j *ai) {
  r af;
  unsigned char *o, aj;
  void *ak;
  k al, am;
  std::type_info *an;
  x *ao;
  bool ap = false;
  ao = &typeid(b);
  while (1) {
    n(o, &al);
    n(o, &am);
    if (al)
      an = t(&af, an || w(an, ao, &ak));
    if (actions ? al : y(&af, al))
      ap = true;
    if (am)
      break;
  }
  if (ap)
    aj = al;
  try {
    std::c();
  } catch (...) {
  }
  i(ai, 1, aj);
}
} // namespace a
```

Command line:
`arc-snps-linux-uclibc-gcc -O2 -S -mcpu=arc700 eh_personality.ii -fpic`

claziss, 2019-11-22:
As a stop gap solution please use `-fno-checking` when compiling the uclibc toolchain. Or configure with `--enable-checking=release` to disable checks.

claziss, 2019-11-26:
New reduced set, now failing for `archs` as well:

```c
template <class> struct a;
struct b {
  static int e;
};
template <typename, template <typename> class> using d = int;
template <> struct a<char> { static char *aa(char *, int, char); };
struct g {
  template <typename> using f = b;
};
struct k : g {
  using ad = d<char, f>;
};
class l {
  struct h {
    h(k::ad);
  } ag;

public:
  l() : ag(0) {}
  ~l();
  int ah();
};
class m {
  class n;
  class o;
};
class m::n {};
class m::o {
public:
  int i();
};
enum al {};
class am {
  typedef al an;
};
template <typename> class j {
  int aq;
  int ar;

public:
  int operator*();
  bool as(j) { return au() == au(); }
  bool au();
};
template <typename c, typename ay> bool operator!=(c p, ay p2) {
  return p.as(p2);
}
template <typename> struct q : m::n {
  char bc;
  char bd[];
};
template <typename> class r : m {
public:
  static o ak;
};
template <typename, typename bf> class bg {
  typedef bf bi;
  template <typename bj> bi bk(bi, bi, am &, am::an &, bj &) const;
};
template <typename> struct s;
template <typename c> struct s<q<c>> {
  q<c> *operator()(m) {
    int bm = r<c>::ak.i();
    m::n *bn;
    return static_cast<q<c> *>(&bn[bm]);
  }
};
template <typename c, typename bf>
template <typename bj>
bf bg<c, bf>::bk(bf p, bf p2, am &, am::an &, bj &t) const {
  typedef q<c> bs;
  s<bs> bt;
  m bl;
  bs bu = *bt(bl);
  char bv;
  int bw, bz, cg;
  bool bx, by, cd = false;
  while (!bx) {
    if (bu.bc)
      break;
    by = ++bz;
    bw = p != p2;
  }
  l cc;
  while (!bx) {
    char *ci = a<c>::aa(bu.bd, bw, bv);
    if (ci)
      break;
    cg = ci - bu.bd;
    if (b::e)
      cd = cg;
    if (p != p2)
      bv = *p;
  }
  bz &&by &&cc.ah();
  if (cd)
    t = cg;
}
template j<char> bg<char, j<char>>::bk(j<char>, j<char>, am &, am::an &,
                                       long &) const;
```

falaleevms, 2020-07-29:
In latest gcc mainline verification (e0633768a1a2efe689e5505b3a95aa949d704b06) launch uclibc toolchain was built without errors.

closed 2020-07-29 by falaleevms