foss-for-mips-arc-processors/gcc #87: Building llvm testsuite failed: ISO C++17 does not allow dynamic exception specifications

falaleevms, 2020-07-29:
Reproducing problem (in ru20 site):

```
$ /slowfs/ru20arcgnu1/gnu_verification/builds/127_28.07.2020_scripts/arc-default-elf32//bin/arc-elf32-g++ -c exception_spec_test.i
/u/svc-arcoss_auto/reference_repositories/source_code/llvm-test-suite/SingleSource/Regression/C++/EH/exception_spec_test.cpp:28:21: error: ISO C++17 does not allow dynamic exception specifications
   28 | void test(bool Int) throw (double) {
      |                     ^~~~~
```

This error occurs during building llvm testsuite.

Attaching [exception_spec_test.i.txt](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/4994487/exception_spec_test.i.txt)

You can also find this file here: `/remote/ru20home1/falaleev/saved/exception_spec_test.i`

This error first was discovered during verification of gcc mainline of following commit: 3cbc0fb39c84ae0a51a9a88649dccd105bf17d6e

On `dev` branch error was discovers during verification of efa751846724497cfdb3935aeec9b485d404a8ea

falaleevms, 2020-10-26:
@claziss any thoughts on this?

claziss, 2020-10-26:
This is an old test and needs to be deprecated. Dynamic exception specifications are deprecated since C++11 standard.

closed 2020-10-26 by claziss