foss-for-mips-arc-processors/gcc #93: ICR during gcc build for hs38 cpu

falaleevms, 2020-12-28:
Toolchain build failed during verification of dev branch (dc816cd4)

Building gcc for hs38 fails with following error for elf toolchains:

```
$ /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/bd-elf32/gcc-stage2/./gcc/xg++ -shared-libgcc -B/slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/bd-elf32/gcc-stage2/./gcc -g -O2 -mcpu=hs38 -c memory_resource.i
during RTL pass: loop2_invariant
/slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/libstdc++-v3/src/c++17/memory_resource.cc: In member function ‘virtual void* std::pmr::unsynchronized_pool_resource::do_allocate(std::size_t, std::size_t)’:
/slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/libstdc++-v3/src/c++17/memory_resource.cc:1455:3: internal compiler error: Segmentation fault
1455 |   }
      |   ^
0xf65a6f crash_signal
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/toplev.c:327
0x13771f8 arc_legitimate_address_p
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/config/arc/arc.c:6737
0xeb40fe memory_address_addr_space_p(machine_mode, rtx_def*, unsigned char)
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/recog.c:1793
0xef6ec2 address_cost(rtx_def*, machine_mode, unsigned char, bool)
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/rtlanal.c:4644
0xd893fc create_new_invariant
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-invariant.c:731
0xd893fc find_invariant_insn
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-invariant.c:1141
0xd893fc find_invariants_insn
       /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-invariant.c:1181
0xd893fc find_invariants_bb
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-invariant.c:1200
0xd893fc find_invariants_body
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-invariant.c:1224
0xd893fc find_invariants
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-invariant.c:1243
0xd89b52 move_single_loop_invariants
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-invariant.c:1949
0xd89b52 move_loop_invariants()
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-invariant.c:2289
0xd85310 execute
        /slowfs/ru20arcgnu1/gnu_verification/arc-default-elf32/gcc/gcc/loop-init.c:530
```

And with following error for linux toolcahins:
```
$ /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/bd-uclibc/gcc-stage2/./gcc/xg++ -shared-libgcc -B/slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/bd-uclibc/gcc-stage2/./gcc -O2 -g -mcpu=hs38 -c floating_to_chars.i
during RTL pass: reload
/slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/libstdc++-v3/src/c++17/floating_to_chars.cc: In function ‘std::to_chars_result std::__floating_to_chars_shortest(char*, char*, T, std::chars_format) [with T = float]’:
/slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/libstdc++-v3/src/c++17/floating_to_chars.cc:1023:3: internal compiler error: in operator[], at vec.h:890
1023 |   }
      |   ^
0x5c99f0 vec<reg_equivs_t, va_gc, vl_embed>::operator[](unsigned int)
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/vec.h:890
0x1377f87 vec<reg_equivs_t, va_gc, vl_embed>::operator[](unsigned int)
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/config/arc/arc.c:914
0x1377f87 arc_secondary_reload
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/config/arc/arc.c:904
0xdb42d1 check_and_process_move
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/lra-constraints.c:1342
0xdb42d1 curr_insn_transform
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/lra-constraints.c:4063
0xdb729e lra_constraints(bool)
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/lra-constraints.c:5138
0xd9eb5a lra(_IO_FILE*)
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/lra.c:2329
0xd50d39 do_reload
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/ira.c:5802
0xd50d39 execute
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/ira.c:5988
```

Attaching .i files
[memory_resource.i.txt](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/5747151/memory_resource.i.txt)
[floating_to_chars.i.txt](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/5747153/floating_to_chars.i.txt)

claziss, 2020-12-28:
Fixed in:

b5ce3ed arc: Make use reg_renumber safe. - Claudiu Zissulescu
47323bd arc: flip if-condition predicates in secondary reload hook - Claudiu Zissulescu

Thanks,

falaleevms, 2020-12-28:
Still cannot build linux toolchain for hs38, getting following error in the same file:

```
$ /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/bd-uclibc/gcc-stage2/./gcc/xg++ -shared-libgcc -B/slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/bd-uclibc/gcc-stage2/./gcc -O2 -g -mcpu=hs38 -c memory_resource.i  
during RTL pass: dse2
/slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/libstdc++-v3/src/c++17/memory_resource.cc: In member function ‘void* std::pmr::__pool_resource::_Pool::try_allocate()’:
/slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/libstdc++-v3/src/c++17/memory_resource.cc:673:5: internal compiler error: in cselib_record_set, at cselib.c:2614
  673 |     }
      |     ^
0xa9b647 cselib_record_set
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/cselib.c:2614
0xaa222d cselib_record_sets
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/cselib.c:2943
0xaa3259 cselib_process_insn(rtx_insn*)
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/cselib.c:3066
0x174ebee dse_step1
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/dse.c:2764
0x174ebee rest_of_handle_dse
        /slowfs/ru20arcgnu1/gnu_verification/arc-archs-linux-uclibc_hs38/gcc/gcc/dse.c:3680
```

claziss, 2020-12-28:
Seems something is broken upstream.

claziss, 2020-12-29:
Please try it again.
1371b3b arc: generate mac(u) insn instead of macd(u) when destination is acc

falaleevms, 2020-12-29:
still getting the same error ([memory_resource.i.txt](https://github.com/foss-for-synopsys-dwc-arc-processors/gcc/files/5751060/memory_resource.i.txt)):

```
$ /slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/bd-uclibceb/gcc-stage2/./gcc/xg++ -shared-libgcc -B/slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/bd-uclibceb/gcc-stage2/./gcc -O2 -g -mcpu=hs38  -c memory_resource.i
during RTL pass: dse2
/slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/gcc/libstdc++-v3/src/c++17/memory_resource.cc: In member function ‘void* std::pmr::__pool_resource::_Pool::try_allocate()’:
/slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/gcc/libstdc++-v3/src/c++17/memory_resource.cc:673:5: internal compiler error: in cselib_record_set, at cselib.c:2614
  673 |     }
      |     ^
0xa9bec7 cselib_record_set
        /slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/gcc/gcc/cselib.c:2614
0xaa2aad cselib_record_sets
        /slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/gcc/gcc/cselib.c:2943
0xaa3ad9 cselib_process_insn(rtx_insn*)
        /slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/gcc/gcc/cselib.c:3066
0x174f62e dse_step1
        /slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/gcc/gcc/dse.c:2764
0x174f62e rest_of_handle_dse
        /slowfs/ru20arcgnu1/gnu_verification/arceb-archs-linux-uclibc_hs38/gcc/gcc/dse.c:3680
```

claziss, 2020-12-30:
It should be fixed:
62a51b2 arc: fix accumulator first register.

falaleevms, 2021-01-12:
Confirm, toolchain was built successfully.

closed 2021-01-12 by falaleevms