foss-for-mips-arc-processors/gcc #143: Option 'mpy-option=w' is not available for 'arc600' CPU

dasilt, 2025-03-17:
Im trying to compile code form ARC600 dual core CPU but I'm unable set GCC to use MPYW and MPYUW instructions.
When calling gcc with `-mcpu=arc600 -mmpy-option=w` getting: `Option 'mpy-option=w' is not available for 'arc600' CPU.`
I know for sure that my processor is supporting these instructions, because alternatively I'm copiling same code with MetaWare compiler with -Xmpy16 specified, and I see binary code containing mpyw and mpyuw instructions which runs on target without any problems.

abrodkin, 2025-03-17:
@kolerov please take a look at this one.

kolerov, 2025-03-27:
@dasilt Unfortunately, GCC for ARC600 does not support 16x16 multiplication instructions (`MPYW` and `MPYUW`).